yuri91/ili9341-rs #2: Display bus support

tarcieri, 2019-04-29:
We were interested in using this crate with the AdaFruit PyPortal. One small problem... the board does not support SPI (out-of-the-box):

https://learn.adafruit.com/adafruit-pyportal/pinouts

> To give you the most data throughput we configure the screen for 8-bit interfacing. That means 8 data lines and a collection of 4 or 5 control lines. If you really want to use the screen in SPI mode, you can do so by soldering closed the SPI jumper and cutting/resoldering the 8/SPI jumper over to the SPI side. That's for advanced users!

...so we could resolder the board to support SPI mode, but it seems like the display bus mode would be ideal, and also ensure that other users of the same board don't have to modify the hardware.

We were looking at the AdaFruit `displayio` CircuitPython package and saw it has two modes:

https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/__init__.html

> - `Display` – Manage updating a display over a display bus
> - `FourWire` – Manage updating a display over SPI four wire protocol

We were curious if you'd be interested in supporting both modes in this crate, as it seems that would be the ideal way for us (and other potential Rust PyPortal) users to leverage the display both out-of-the-box, and with the best performance.

yuri91, 2019-05-02:
Hi!
I am interested in having support for the display bus mode in this crate.
I don't have a device with that interface, but if you manage to get a working implementation I will accept a PR.

sajattack, 2019-05-05:
Started working on this here https://github.com/sajattack/ili9341-rs/tree/8bit-parallel

yuri91, 2020-02-12:
@sajattack what are your thoughts on this issue after https://github.com/yuri91/ili9341-rs/pull/7 and https://github.com/yuri91/ili9341-rs/pull/7 ?

sajattack, 2020-02-12:
Feel free to close it.

yuri91, 2020-02-12:
oh, actually I meant to tag @tarcieri , who opened the issue

tarcieri, 2020-02-13:
I can try to test it out at some point, but otherwise it looks like what's needed

yuri91, 2020-02-14:
Ok. I am going to close this. Feel free to reopen or file a new issue if something is missing or not working.

closed 2020-02-14 by yuri91