yuri91/ili9341-rs #3: Display only supports black and white?

therealprof, 2019-05-02:
So I was just wondering why I only get black and white out of the display and then I noticed this:

https://github.com/yuri91/ili9341-rs/blob/54d5381ec40b5e304dc46dad3174bbe0d3fcb2b7/src/lib.rs#L303-L307

which looks kind of wrong to me 🤔

yuri91, 2019-05-02:
If you use the raw methods (`draw_raw` and `draw_iter`) you can have all the colours you want in the rgb565 space.
That code is part of the [newly merged PR](https://github.com/yuri91/ili9341-rs/pull/1) that updates the optional support of embedded_graphics to version 0.4. The first version of embedded_graphics only supported black and white and the PR just updated the minimum to make it compile.

I plan to extend the functionality in the near future.

It would be also interesting to see if embedded_graphics can support efficient drawing (setting the bounding box of the area to update at the beginning, and only send the colours to the display, not the coordinates for every pixel)

therealprof, 2019-05-02:
Actually I think that's what it does... I do have problems converting the data to use `draw_iter` though:

```
   |
57 |         lcd.draw_iter(
   |             ^^^^^^^^^ expected struct `embedded_graphics::drawable::Pixel`, found u16
   |
   = note: expected type `embedded_graphics::drawable::Pixel<_>`
              found type `u16`
```

yuri91, 2019-05-02:
[draw_iter](https://github.com/yuri91/ili9341-rs/blob/54d5381ec40b5e304dc46dad3174bbe0d3fcb2b7/src/lib.rs#L212) expects an iterator of `u16`.
It does not use the `embedded_graphics` types, but expects the `u16` to be the color of a pixel in rgb565 format (that is what the device natively handles)

therealprof, 2019-05-02:
My bad, you're right. Clearing a screen or rendering an image works somewhat fast now. Readout would be great. ;)

yuri91, 2019-12-19:
A recent [PR](https://github.com/yuri91/ili9341-rs/pull/4) added color support and the latest version of embedded-graphics. I am closing this.

closed 2019-12-19 by yuri91