yuri91/ili9341-rs #9: New release?

Disasm, 2020-02-26:
Since #7 and #8 have landed, do you mind releasing a new version of this crate?

yuri91, 2020-02-26:
Yeah, a new release makes sense, but I was waiting for embedded-graphics to release 0.6 (https://github.com/jamwaffles/embedded-graphics/projects/6), since right now we depend on 0.6.0-alpha.2.

Maybe I should release an alpha version as well?

jamwaffles, 2020-02-26:
We release [0.6.0-beta.1](https://crates.io/crates/embedded-graphics/0.6.0-beta.1) a few days ago. I guess you could release a beta of this crate?

Assuming nothing major comes up, there should be no more breaking changes before the release of 0.6.0 proper.

yuri91, 2020-02-26:
Seems reasonable to me. I will try to update to 0.6.0-beta.1 of embedded-graphics and then release a 0.3.0-beta.1 of this crate to crates.io

jamwaffles, 2020-02-26:
Sounds good to me. That's what I've been doing with SSD1306 etc

yuri91, 2020-03-01:
I just implemented the new embedded_graphics interface and published 0.3.0-beta.1 on crates.io!

Let me know if there are issues!

closed 2020-03-01 by yuri91