yuri91/ili9341-rs #12: Broken drawing of some rectangles using embedded-graphics

rfuest, 2020-04-27:
The accelerated `draw_rectangle` method in the `DrawTrait` impl doesn't work correctly for rectangles with only a stroke applied (e.g. `PrimaryStyle::with_stroke(Rgb565::RED, 10)`), because the embedded-graphics iterator skips the transparent pixels.

I would suggest to simply fall back to unaccelerated drawing if `fill_color.is_none()` at the moment. We plan to change the `DrawTarget` trait soon, which should make it easier to implement it correctly. If you have any suggestions how we could improve the interface, from a driver author perspective, the discussion about the changes is in https://github.com/jamwaffles/embedded-graphics/issues/330.

yuri91, 2020-05-04:
Thanks for pointing out this bug!
I will also try to think if I have suggestions for the new trait!

yuri91, 2022-04-18:
closing since embedded-graphics changed a lot since this was opened

closed 2022-04-18 by yuri91