yuri91/ili9341-rs #13: Add some examples

pepijndevos, 2020-05-04:
It took me two days to get Rust and this library working on my STM32F4DISCOVERY, so it would be really amazing to have some examples. I'll share the code I ended up with, but I'm not sure about the correct setup to add it as a proper example. My project is based on https://github.com/stm32-rs/stm32f407g-disc with the following main file

```rust
#![no_main]
#![no_std]

use panic_halt as _;

use stm32f407g_disc as board;

use crate::board::{
    hal::stm32,
    hal::spi::{Mode, Phase, Polarity, Spi},
    hal::{delay::Delay, prelude::*},
};

use cortex_m::peripheral::Peripherals;

use cortex_m_rt::entry;

use embedded_graphics::{
    fonts::{Font8x16, Text},
    pixelcolor::Rgb565,
    prelude::*,
    primitives::{Circle, Rectangle},
    style::{PrimitiveStyle, TextStyle},
};

use ili9341::{Ili9341, Orientation};

#[entry]
fn main() -> ! {
    if let (Some(p), Some(cp)) = (stm32::Peripherals::take(), Peripherals::take()) {

        let rcc = p.RCC.constrain();
        let gpioa = p.GPIOA.split();

        let clocks = rcc
            .cfgr
            .use_hse(8.mhz())
            .sysclk(48.mhz())
            .pclk1(24.mhz())
            .freeze();

        // Configure pins for SPI
        let sck = gpioa.pa5.into_alternate_af5();
        let miso = gpioa.pa6.into_alternate_af5();
        let mosi = gpioa.pa7.into_alternate_af5();

        let spi_mode = Mode {
            polarity: Polarity::IdleLow,
            phase: Phase::CaptureOnFirstTransition,
        };

        let spi = Spi::spi1(
            p.SPI1,
            (sck, miso, mosi),
            spi_mode,
            16.mhz().into(),
            clocks,
        );
        // Get delay provider
        let mut delay = Delay::new(cp.SYST, clocks);

        let mut display = Ili9341::new_spi(
            spi,
            gpioa.pa4.into_push_pull_output(),
            gpioa.pa3.into_push_pull_output(),
            gpioa.pa2.into_push_pull_output(),
            &mut delay
        ).expect("display init failed");

        display.set_orientation(Orientation::LandscapeFlipped);

        let c = Circle::new(Point::new(20, 20), 50)
            .into_styled(PrimitiveStyle::with_fill(Rgb565::RED));
        let t = Text::new("Hello Rust!", Point::new(20, 16))
            .into_styled(TextStyle::new(Font8x16, Rgb565::GREEN));
        let r = Rectangle::new(Point::new(0, 0), Point::new(320, 240))
            .into_styled(PrimitiveStyle::with_fill(Rgb565::BLACK));

        r.draw(&mut display).expect("draw failed");
        c.draw(&mut display).expect("draw failed");
        t.draw(&mut display).expect("draw failed");
    }

    loop {
        continue;
    }
}
```

yuri91, 2020-05-04:
Hi!

The issue with examples is that I have to assume a particular board for the example itself, and most of the code would be setting up the specific board and its peripherals, more that using the library.

That said, having at least one (stm32f01? raspberry pi?) is probably warranted

pepijndevos, 2020-05-04:
An alternative could be to add ILI9341 examples to upstream HAL crates. The fact that the STM32F4 HAL crate did not have an SPI example was pretty much day two of my struggles. I had to adapt it from the F3 crate.

But yea, assuming the HAL crate has an SPI example, I'd just add an example for whichever board you have at home that is reasonably popular.

maxdignan, 2021-10-12:
I think I'd really benefit from more examples. Thanks @pepijndevos for sharing your's! I'm working on an F3-Discovery with a 3.5" TFT Touchscreen Breakout [adafru.it/2050](http://adafru.it/2050) . Running into issues getting it to do anything more than light up. Any suggestions @yuri91 ?

```
#![feature(alloc_error_handler)]
#![no_main]
#![no_std]

extern crate alloc;

use alloc::boxed::Box;
use alloc::vec;
use alloc::vec::Vec;
use cortex_m::Peripherals;
use cortex_m::delay::Delay;
use cortex_m::peripheral::SYST;
use cortex_m::peripheral::syst::SystClkSource;
use cortex_m::prelude::{_embedded_hal_blocking_delay_DelayMs, _embedded_hal_blocking_spi_Write};
use f3::hal::rcc::Rcc;
use core::borrow::Borrow;
use core::marker::PhantomData;
use core::ops::Deref;
use core::{alloc::Layout, pin::Pin};
use core::fmt::Write;
use core::panic::PanicInfo;
use cortex_m::{asm::nop, iprintln};
use cortex_m_rt::{entry, exception, ExceptionFrame};

use embedded_hal::{digital::v2::OutputPin, spi::{Mode, Phase, Polarity}, blocking::delay::DelayMs, };
use embedded_sdmmc::{DirEntry, TimeSource, Timestamp};
use f3::hal::{gpio::{gpioa::{PA0, PA1, PA4, PA5, PA6, PA7}, gpioe, GpioExt, Output, PushPull, AF5},  i2c::SdaPin, prelude::_stm32f30x_hal_flash_FlashExt,  rcc::{Clocks, RccExt, APB2}, spi::{MisoPin, MosiPin, SckPin, Spi}, stm32f30x::{self, FLASH, GPIOA, RCC, SPI1 }, time::Hertz};
use heapless::String;
use panic_itm; // panic handler

use embedded_graphics::{
  fonts::{Font8x16, Text},
  pixelcolor::Rgb565,
  prelude::*,
  primitives::{Circle, Rectangle},
  style::{PrimitiveStyle, TextStyle},
};

// use embedded_graphics_core::{primitives::rectangle::Rectangle, draw_target::DrawTarget, pixelcolor::Rgb565, geometry::{Size, Point}};

use ili9341::{Ili9341, Orientation};
use display_interface;

// use state_mgmt;

use alloc_cortex_m::CortexMHeap;
// this is the allocator the application will use
#[global_allocator]
static ALLOCATOR: CortexMHeap = CortexMHeap::empty();

const HEAP_SIZE: usize = 1024 * 40; // 40 KB .. the RAM size on f3discovery

// struct my_delay;

// impl DelayMs<u16> for my_delay {
//   fn delay_ms(&mut self, ms: u16) {
//     delay(ms.into())
//   }
// }

struct SPIWrapper {
  spi: Box<Spi<SPI1, (PA5<AF5>, PA6<AF5>, PA7<AF5>)>>,
}

impl display_interface::WriteOnlyDataCommand for SPIWrapper {
  fn send_commands(&mut self, cmd: display_interface::DataFormat<'_>) -> Result<(), display_interface::DisplayError> {
    match cmd {
[... 150 lines cut ...]
    }

    loop {
        cortex_m::asm::nop();
    }
}

// define what happens in an Out Of Memory (OOM) condition
#[alloc_error_handler]
fn alloc_error(_layout: Layout) -> ! {
    loop {
        cortex_m::asm::bkpt();
        cortex_m::asm::nop();
    }
}

#[exception]
fn HardFault(ef: &ExceptionFrame) -> ! {
    loop {
        cortex_m::asm::bkpt();
        cortex_m::asm::nop();
    }
}

```

plaes, 2021-11-14:
@maxdignan You are using an old and already removed API. Here's a barebones skeleton example:
```rust
use display_interface_spi::SPIInterface;
use embedded_time::rate::Extensions;
use ili9341::{Ili9341, Orientation};

use embedded_graphics::{
    pixelcolor::Rgb565,
    prelude::*,
    primitives::{PrimitiveStyle, Rectangle},
};

fn main() {
    /// ... most of the peripheral setup is dropped...

    let cs = pins.gpio5.into_push_pull_output();
    let dc = pins.gpio11.into_push_pull_output();
    let reset = pins.gpio10.into_push_pull_output();

    let spi = hal::spi::Spi::<_, _, 8>::new(pac.SPI0);
    
    let spi = spi.init(
        &mut pac.RESETS,
        clocks.peripheral_clock.freq(),
        16_000_000u32.Hz(),
        &embedded_hal::spi::MODE_0,
    );

    let spi_iface = SPIInterface::new(spi, dc, cs);

    let mut display = Ili9341::new(
        spi_iface,
        reset,
        &mut delay,
        Orientation::Portrait,
        ili9341::DisplaySize240x320,
    ).unwrap();

    display.clear(Rgb565::BLACK).unwrap();
    let on = PrimitiveStyle::with_fill(Rgb565::RED);

    Rectangle::new(Point::new(10, 20), Size::new(10, 10))
        .into_styled(on)
        .draw(&mut display).unwrap();

    loop {}
}
```

maxdignan, 2021-12-14:
@plaes Thank you so much! This was super helpful to me!

yuri91, 2022-04-18:
A basic example was added in https://github.com/yuri91/ili9341-rs/pull/32, so I am closing this

closed 2022-04-18 by yuri91