yuri91/ili9341-rs #15: Gpio8Interface data_pins issues

tarcieri, 2020-05-16:
I'm trying to wire up `Gpio8Interface` and it seems trickier than it needs to be.

I think my main concerns are:

- Display bus pins are borrowed, other pins are owned
- Display bus pins use `dyn OutputPin`, other pins are monomorphized

I'm dealing with a type where all of the pins are owned:

https://docs.rs/pyportal/0.4.0/pyportal/pins/struct.Display.html

I'd like to just hand off ownership of all of the pins, including the display bus pins, to the display driver. It seems right now I can only hand over some of them, and the others need to be borrowed.

I think it might make sense to have some sort of display bus type where all of the pins are generic instead of the current:

https://docs.rs/ili9341/0.3.0/ili9341/gpio/struct.Gpio8Interface.html#method.new

```rust
data_pins: &'a mut [&'a mut dyn OutputPin<Error = PinE>; 8],
```

That'd also eliminate any concerns around lifetimes.

yuri91, 2020-05-17:
Hi!

This interface was added by @flaminggoat . I personally don't have a use case for the gpio interface, so I am not sure if he had a particular use case in mind for implementing it like it is now.

In principle I have nothing against changing it. I would wait a bit to see if somebody that uses the current interface has some thoughts on this.

tarcieri, 2020-05-17:
If @flaminggoat is ok with it, I can send a PR to change it to something like the above. I think it'd really help with simplifying pin ownership.

yuri91, 2020-09-01:
Since https://github.com/yuri91/ili9341-rs/pull/16 landed, the actual display interface is implemented by https://github.com/therealprof/display-interface (and friends).

So I am going to close this

closed 2020-09-01 by yuri91