yuri91/ili9341-rs #27: Only embedded-graphics-core should be required

jamwaffles, 2021-06-08:
I could be missing something here, but it shouldn't be necessary to implement integrations for both embedded-graphics and embedded-graphics-core - the intent with the crate split in e-g 0.7 is that only the latter is required.

Are both implementations in the crate to support older versions of e-g perhaps?

jamwaffles, 2021-06-08:
Oh, I just found https://github.com/yuri91/ili9341-rs/pull/26#issuecomment-840666446. I'll rename this ticket!

yuri91, 2021-06-08:
Yes, the plan is to support only 0.7 not that is out. I will do it this weekend probably

yuri91, 2021-06-10:
I removed support for embedded-graphics 0.6.x.
I also bumped the version to 0.5.0 and pushed it to crates.io

closed 2021-06-10 by yuri91

jamwaffles, 2021-06-10:
Awesome! Great to see the new version being used :)