yuri91/ili9341-rs #36: When I attach an Adafruit ST7789, it acts like the color space is b3r2g3, and is bit flipped

mutantbob, 2022-06-14:
I purchased a https://www.adafruit.com/product/4311 .  I can run the https://github.com/adafruit/Adafruit_ILI9341/tree/master/examples/pictureEmbed arduino sketch and it mostly works (it malfunctions a little when it tries to draw at x<0).

When I try to use the ili9341-rs library, the `draw_raw_iter()` function does not work like I expected.  After much fiddling around, I have concluded that the MSB of the u16 pixel is irrelevant, and the LSB seems to be interpreted as bbbrrggg with the bits inverted (0 is white, 0xff is black).

Is the ili9341 interface applicable to this piece of hardware, or is it a different interface that Adafruit jammed into the library?  Maybe there is an SPI configuration command that their C++ library sends that the rust one overlooked?

If it helps to look at my source code: 
https://github.com/mutantbob/ili9341-exp-2
https://github.com/mutantbob/ili9341-exp-2/blob/0f86310f63b3cbcc732e0428ba829b0ab8ccfae3/src/main.rs#L218

mutantbob, 2022-06-14:
Looking at the adafruit example, the dragon flag displays on my screen as a green dragon over a black-over-blue flag. (per fess black and blue?) .

If I run the pattern through a rust function to convert the rgb565 into g3r2b3, it shows up as the flag of wales (flipped horizontally) ; a red dragon passant to sinister on a green and white field.

So this problem seems to affect both libraries.  The question becomes: how do I rig it to process 16-bit color?

mutantbob, 2022-06-14:
I have opened an issue against the Adafruit library.  Hopefully their fix will enable us to adjust the rust crate as well.
https://github.com/adafruit/Adafruit_ILI9341/issues/80

yuri91, 2022-06-14:
Do you have a datasheet for it? I can't find it on the product page

mutantbob, 2022-06-15:
You are correct.  I have converted to ST7789, and suffer the same color space problem.  I will continue experimenting, but my problem is not an `ili9341-rs` problem.