yuri91/ili9341-rs #42: Question: how to read touch screen input ?

ValouBambou, 2024-02-14:
Hi, I'm kind of new to embedded programming and I managed to use embassy + embedded_graphics to connect a Raspberry Pi Pico to an ili9341 for displaying stuff, but I can't find how to read the touch screen input using this library. From what I understand the touch part is connected through a different SPI bus, thus I should send command and data to this bus, but I don't understand what to send in order to read the optional x y coordinates of the touched area on the touch screen. If the feature is missing, I could implement it.

yuri91, 2024-02-14:
I believe that the touch screen, as you say, is on a different spi device. And the ili9341 devices that I have do not even have it.

So I think that it's out of scope for this library to implement it. Another library can implement the touch screen and be used together with this for the display functionality.

ValouBambou, 2024-02-14:
This makes sense, I apologize for my dumb question you're right the touch screen is a different device that's included in the display, it's a XPT2046 controller so now I can reuse the other code from embassy example that's using this exact same touching device. Thanks for your insight !

yuri91, 2024-02-14:
No worries and good luck!