yuri91/ili9341-rs #43: Performance improvements without DMA

ValouBambou, 2024-03-06:
Hi, I was doing a GPS like application on a Raspberry Pi Pico using embassy and embedded_graphics, and I encounter some performances issues. 

## First measurement

Indeed, clearing the screen took 263 ms on average and rendering the first frame of my app (rendering 20 lines) took around 340ms. These time are a bit too high for reaching the goal of 60 fps. So I planned to use DMA, but before I was wondering if there are some tweaks to improve the driver.  

## Considered improvements

My first idea when looking at the code was to avoid some slice to iterator conversions, e.g. in `command`, so based on that I try some tweaks:

- In `command` use `DataFormat::U8` Instead of the `DataFormat::U8Iter.
- Add private method `write_slice` specialized version of `write_iter` but with `DataFormat::U16` instead of `DataFormat::U16BEIter` (I hope the endianness is not an issue).
- Use this new method in public method `draw_raw_slice`.

With this simple idea, I got a **speedup around 18%** on the rendering of my first frame (281 ms instead of 338 ms) and just a consistent save of 1ms for clearing the screen. I'm looking to push the improvement a bit more before trying DMA and I might need some help.

I was wondering about making a special private_primitive `draw_pixel` in order to replace the call to `draw_raw_iter` in `WriteOnlyDataCommand::draw_iter` which seems to be overkill. Furthermore, I'm not sure of how to write it because of the command sent by `set_window` (it will send 2 times x and 2 times y).  Is this step necessary ? Could we just send 1 time instead of 2 ? I don't really understand the details of the protocol used for the communication with the ili9341.

Once all my changes are done and their impact has been measured, I will open a PR.

yuri91, 2024-03-06:
One thing that can help a lot is to try to increase the frequency of the SPI bus.
You can play with it and increase it until the device can't keep up anymore.

Looking at your numbers I think that you are indeed limited by the spi speed, because as you say you get a good speedup for the first frame, but then not so much anymore, suggesting that the bottleneck at the steady state is not the cpu. 

I will add specific feedback about the PR over there.

Thanks for taking the time to experiment, report here and do a PR!

ValouBambou, 2024-03-06:
I tried to increase the frequency, but now the `clear` method is taking 85 ms and the first frame 171 ms. While when I undo my changes the clear screen takes the same amount of time, but the first frame took 242 ms, so there is still is a big 30% difference even when the frequency is high. 

Also, when I talk about the first frame it is just to simplify, the time for the next frames is also reducing.

PR #44 (2024-03-06): Performance improvement of `draw_slice` and `draw_iter`
This implements the proposition to fix #43
closed 2024-03-06 by yuri91