yuri91/ili9341-rs #47: Version 0.6.0 not working with SPIInterface anymore

RedPixlFox, 2024-06-25:
Hi, I was recently experimenting with my stm32f429 board and discovered (a copy of) this crate [here](https://github.com/chenx97/stm32f429_lcd_demo.git).
Today I tried to implement the newest version (0.6.0) in my project to use the lcd, but the SPIInterface::new() doesnt work like it does in the [example](https://docs.rs/ili9341/0.6.0/ili9341/#usage) in the docs. This is because the newest version of [display-interface-spi](https://docs.rs/display-interface-spi/0.5.0/display_interface_spi/struct.SPIInterface.html) no longer takes the "chip-select pin" as argument (like it did in the older [version](https://docs.rs/display-interface-spi/0.4.1/display_interface_spi/struct.SPIInterface.html)) and therefore no longer implements the [SpiDevice](https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html) trait from the embedded-hal crate which is needed for the [WriteOnlyDataCommand](https://docs.rs/display-interface-spi/0.5.0/display_interface_spi/struct.SPIInterface.html#impl-WriteOnlyDataCommand-for-SPIInterface%3CSPI,+DC%3E) trait (but the Ili9341::new(...) function requires the "interface" argument to implement that trait)

I would really appreciate, if someone could tell me what to do for going around that 👍

paullj, 2024-09-07:
Getting the same issue, did you manage to resolve it?

malteki, 2024-09-07:
No sadly not, but the other version/copy (linked in my first message) works perfectly fine

malteki, 2024-09-07:
Wait why did I close it 🥲

No I couldn't fix it, but the other version/copy (linked in my first message) works perfectly fine

paulc, 2024-10-21:
It it helps the following code works - you need to use one of the SPIDevice implementations in [embedded-hal-bus](https://docs.rs/embedded-hal-bus) to create a instance that implements the SpiDevice trait (the easiest is [ExclusiveDevice](https://docs.rs/embedded-hal-bus/latest/embedded_hal_bus/spi/struct.ExclusiveDevice.html) which encapsulates CS)

```
    // ILI9341 LCD
    let lcd_clk = gpiob.pb13.into_alternate();
    let lcd_miso = NoMiso::new();
    let lcd_mosi = gpiob.pb15.into_alternate().internal_pull_up(true);

    let mode = Mode {
        polarity: Polarity::IdleLow,
        phase: Phase::CaptureOnFirstTransition,
    };

    let lcd_dc = gpiob.pb0.into_push_pull_output();
    let lcd_cs = gpiob.pb1.into_push_pull_output();
    let lcd_reset = gpiob.pb2.into_push_pull_output();
    let mut lcd_backlight = gpiob.pb12.into_push_pull_output();
    
    let spi2 = Spi::new(
        dp.SPI2, 
        (lcd_clk, lcd_miso, lcd_mosi),
        mode,
        24.MHz(),
        &clocks,
    );  
    let spi_delay = dp.TIM11.delay_us(&clocks);
    let spi_device = ExclusiveDevice::new(spi2, lcd_cs, spi_delay).unwrap();
    let display_if = SPIInterface::new(spi_device, lcd_dc);
        
    log::info!("Configuring Display");
    
    // Using Ili9341 driver
    let mut display = Ili9341::new(
        display_if,
        lcd_reset,
        &mut delay,
        Orientation::Landscape,
        DisplaySize240x320,
    )
    .unwrap();
```

malteki, 2024-11-04:
Thank you so much. I'll try that, next time I'm working on that project

closed 2024-11-04 by malteki